chipsalliance/caliptra-sw #2273: [Caliptra 2.1][RFC] Make Path Length Constraint vendor-configured

jhand2, 2025-06-26:
Caliptra DICE certificates currently have a pathLenConstraint (in the BasicConstraints extension) of [5](https://github.com/chipsalliance/caliptra-sw/blob/main-2.x/x509/build/build.rs#L56).

With the introduction of DPE ExportCDI, this potentially limits the SoC's ability to control how many certs are in the chain.

I recommend we change the constraint to be vendor-configured.

1. Remove the PathLenConstraint from IDevID and LDevID certificates
2. Add a vendor-signed field to the ImageHeader to define the max path length for FMC Alias and below
3. Update certs to use the vendor-defined value
    * FMC Alias = vendor_pathlen
    * RT Alias = vendor_pathlen - 1
    * DPE ECA (returned from DeriveContext) = vendor_pathlen - 2

Security justification: Exported DPE CA certificates are only available for PL0 callers. This means the vendor is in full control over how many certs in the chain can be minted (via vendor-signed field) and when/how those certs are minted at runtime (via vendor-signed PL0 code).

jhand2, 2025-06-26:
FYI @varuns-nvidia @mhatrevi @xiaoyuruan

mhatrevi, 2025-12-10:
LGTM