chipsalliance/caliptra-sw #2277: PCIe Timing Requirements and Caliptra Boot Time

sunguk-bin-fad, 2025-06-27:
@varuns-nvidia, @jhand2 

I have some questions regarding PCIe timing requirements and their interaction with Caliptra boot time.
My understanding of PCIe timing, based on the [OCP streaming boot document](https://docs.google.com/document/d/1cjdgcKgOzcug5bBoK6k2Mw2mvsQJElp8bs0ec_xLZHc/edit?tab=t.0) and discussions like [Re: Peripheral attestation timing](https://ocp-all.groups.io/g/OCP-Security/message/255?p=%2C%2C%2C20%2C0%2C0%2C0%3A%3Arecentpostdate%2Fsticky%2C%2CPERST%2C20%2C2%2C0%2C75883325), is illustrated below: (Please correct me if anything is wrong)

Pwr on    ----------------      PERST# -- Detect  ---------  CFG ready     ------------------------->         CFG ready
  |------------100ms------------|--20ms--|-------80ms--------|---(can be deferred with CRS? and +900ms) --|

In essence, the SoC ROM must launch the SoC FMC, and the SoC FMC must then complete PCIe configuration within 120ms from power-on. This implies the SoC ROM needs to launch the SoC FMC within roughly 100ms, leaving about 20ms for FMC's PCIe configuration. 
My primary concern is that our estimated Caliptra boot time of 80ms(targetting 300MHz) significantly impacts this tight window. If Caliptra's boot consumes 80ms, it leaves only 20-30ms for the SoC ROM to launch the FMC, and for the FMC to complete PCIe configuration.

Furthermore, regarding SoC FMC measurements:
- If the SoC FMC is not hitless-updatable, stash_measurements is available and it's provided by Caliptra ROM.
- However, if the SoC FMC is hitless-updatable, stash_measurement is unavailable. In this scenario, we would need to use DPE DeriveContext, which requires Caliptra RTFW. This would mean consuming the initial 80ms for Caliptra boot, plus an additional estimated 40ms for DeriveContext, pushing us well beyond the 120ms limit.
(Background: [stash measurement when exceeding the max number of DPE contexts · Issue #2075 · chipsalliance/caliptra-sw](https://github.com/chipsalliance/caliptra-sw/issues/2075))

This leads me to a few core questions about Caliptra's design intention:
1. Does Caliptra assume that the SoC ROM does not wait for Caliptra's runtime ready, effectively allowing Caliptra's boot time to be ignored in this critical path? (calling 'synchronous' FW_LOAD still consumes 40ms)

2. From this perspective, is the design expectation that the SoC FMC must not be hitless-updatable to meet these stringent PCIe timing requirements?

Any clarification on these points would greatly help us integrate Caliptra into our designs and ensure PCIe compliance.

sunguk-bin-fad, 2025-06-27:
Since the SoC FMC will typically initialize PCIe, the time it takes from power-on until the SoC FMC loads seems critical. I'm curious about the expected time for the SoC ROM to initialize Caliptra (e.g., programming the fuses) and communicate with Caliptra (e.g. Caliptra FW load, stash measurement) Has the Caliptra community intended a specific duration for this process? (or are there requirements regarding minimum Caliptra operating frequency, etc.?) If so, this information would be helpful in estimating the actual time available to the SoC ROM and the time available until the SoC FMC initializes PCIe.

varuns-nvidia, 2025-06-27:
Adding @amullick007 to provide simulation data.
Adding @bharatpillilli to consider documenting bounds in the integration FD.

bharatpillilli, 2025-06-27:
Real world application would be the following - In the world of streaming boot and for other security reasons, we plan to use eSFI or similar solutions (using some existing platform capabilities). In other words, 100ms becomes irrelevant for the above scenario. 

If you have customers that require 100ms and streaming boot, then you can implement streaming boot over USB as your own solution that will make it faster or implement a multi lane I3C.

Also the reality of existing systems is that DDR training takes quite long (in seconds) and device enumeration is well after that from system boot view.

And this whole 100ms of PCIe spec needs to be addressed architecturally using eSFI because it is a random number that got picked way back when devices were much simpler.

sunguk-bin-fad, 2025-06-28:
@bharatpillilli 
1. If an SoC supports streaming boot, is it correct to expect that the early-fw will complete PCIe configuration before loading Caliptra?

2. If an SoC doesn't support streaming boot and instead the SoC's FMC initialize PCIe, should this be interpreted to mean that Caliptra could occupy critical time, and then manufacturers need to make individual considerations for this?

bharatpillilli, 2025-07-30:
> [@bharatpillilli](https://github.com/bharatpillilli)
>
> 1. If an SoC supports streaming boot, is it correct to expect that the early-fw will complete PCIe configuration before loading Caliptra?
> 2. If [...]

For Q1: It is SOC architecture specific question; The answer falls back to what I was stating earlier about eSFI or using USB as the path if streaming boot + 100ms is required by some customer of yours. But to simply answer, SOC dictates what would be part of that early fw component

For Q2: This goes back to the question of above timings for Caliptra 2.0 when SPI is used; which would be the above time + auth of SOC manifest + verification/auth of SOC image(s). Since you opened another issue in a different repo, let the time be provided there

closed 2025-07-30 by sunguk-bin-fad

activebitbit, 2026-08-16:
> [@varuns-nvidia](https://github.com/varuns-nvidia), [@jhand2](https://github.com/jhand2)
>
> I have some questions regarding PCIe timing requirements and their interaction with Caliptra boot time. My [...]

hello @sunguk-bin-fad @bharatpillilli 
1. now, your system, SOC ROM must push caliptra FW and wait done?  Can we push caliptra fw in SOC FMC?
2. if SOC ROM do not need wait caliptra done,  if SOC FMC WDT want to generate warm reset to caliptra, caliptra will boot failed, so WDT generate cold reset to caliptra? 
3. if SOC ROM need wait caliptra FW done, if SOC ROM push caliptra FW failed, caliptra will hang, SOC want to reset(caliptra warm reset？cold reset) to restore system ，it not can fix . if this case. we must cold reset to restore the system？