chipsalliance/caliptra-sw #2286: Add compile time checks for mutually exclusive features

nquarton, 2025-07-01:
Some features are mutually exclusive, such as "fmc" and "runtime". There should be compile-time checks to ensure these mutually-exclusive features aren't getting turned on at the same time due to feature unification in rust.

zhalvorsen, 2026-02-21:
I gave it a brief test by adding the following to `caliptra_common` and `caliptra-drivers`. Both have compile failures for all three cases. We will need to dig a little deeper to figure out how they are all getting unified.

```Rust
#[cfg(all(feature = "rom", feature = "fmc"))]
compile_error!("Features 'rom' and 'fmc' cannot be enabled at the same time.");

#[cfg(all(feature = "rom", feature = "runtime"))]
compile_error!("Features 'rom' and 'runtime' cannot be enabled at the same time.");

#[cfg(all(feature = "fmc", feature = "runtime"))]
compile_error!("Features 'fmc' and 'runtime' cannot be enabled at the same time.");
```

swenson, 2026-02-25:
@zhalvorsen I dug in a little, and there are two problems: clippy and some workspace builds cause feature flag unification.

If I add some additional clauses for those, I was able to get everything to pass, I believe. See #3385.