chipsalliance/caliptra-sw #2366: question on storing CA certificate signature

sanyogkale, 2025-08-12:
Hi All,

Whenever caliptra receives CSR request, it will append the signature algo and signature value as part of CSR request as shown below. 

Signature on X.509 certificate for device identity public key DevIKpub, this is created outside the platform and certificate signature is provisioned into platform.

Can someone confirm where this signature is stored? HSM eFuse, SoC eFuse or some NVM like Flash? Or is it left upto manufacturer for storage location?

**Certificate Request:**
    Data:
        Version: 1 (0x0)
        Subject: CN=Caliptra 1.0 IDevID/serialNumber=7FFEDEF6DCD6697BD0F381AA37B58112F24BDDB5550D2C791B5798AD202A0351
        Subject Public Key Info:
            Public Key Algorithm: id-ecPublicKey
                Public-Key: (384 bit)
                pub:
                    04:31:8c:e7:cc:f1:0d:d3:2f:48:1c:3d:76:fc:64:
                    ba:21:5b:54:d5:c8:60:ad:c7:46:2f:22:32:1c:af:
                    26:47:43:9c:59:cc:ac:18:b2:ec:40:62:2f:18:84:
                    ed:14:23:cb:8a:04:05:6b:1e:8c:71:5d:8c:98:63:
                    f2:f6:fb:71:da:de:33:46:3a:62:d2:1b:af:ae:2e:
                    6d:d8:3e:3c:93:51:dc:fa:0a:0f:de:36:4f:ba:5d:
                    14:b9:ce:6d:da:16:9c
                ASN1 OID: secp384r1
                NIST CURVE: P-384
        Attributes:
            Requested Extensions:
                X509v3 Basic Constraints: critical
                    CA:TRUE, pathlen:5
                X509v3 Key Usage: critical
                    Certificate Sign
                2.23.133.5.4.4: 
                    0....................
    **Signature Algorithm: ecdsa-with-SHA384
    Signature Value:
        30:66:02:31:00:a4:30:2b:92:9d:b5:d3:6e:ac:90:73:8f:df:
        2e:da:ed:65:ae:40:07:a7:65:fa:20:fb:ff:26:f8:74:ad:38:
        86:2d:5a:e3:73:e6:d8:e5:86:3b:31:71:5f:f8:db:82:ab:02:
        31:00:ae:da:6c:36:13:2f:9d:17:de:c2:e2:a4:8f:82:fd:5a:
        38:e9:f0:4f:a5:9f:95:39:28:0c:ac:ee:92:b8:2c:f7:1d:96:
        77:15:10:4e:12:5b:6d:3b:5d:ef:5d:da:f7:c2**

sunguk-bin-fad, 2025-08-14:
For Caliptra1.x, this may help you. 
https://github.com/chipsalliance/Caliptra/blob/main/doc/caliptra_1x/Caliptra.md#idevid-certificate
- The SoC is **recommended** to store the IDevID certificate signature in **fuses** and the IDevID certificate template in the firmware image. 
- GET_IDEV_CERT to return the certificate given a to-be-signed (TBS) payload and the certificate signature from fuses.

jhand2, 2025-09-04:
> When you say IDEVID certificate template in the firmware image, how does the requester for eg. BMC knows about the certificate template. do we share or store some template certificate format in BMC?

This is up to the SoC integration, there are several ways it can be done and the right choice depends on product & customer needs.

A few ways it can be done:

* Store the entire certificate somewhere off-chip (like BMC or CPU flash) and load the certificate at boot time.
* Store the signature in additional fuses (outside Caliptra). Reconstruct the certificate using a template known to the SoC (as mentioned above).

> are all fields in certificate mandatory?

* The requested extensions from the CSR should be included in the certificate
* The subject name _must_ match the CSR for path building to work

Additional extensions are up to the SoC integrator.

Are there any specific fields you were concerned about?