chipsalliance/caliptra-sw #2432: Caliptra Core 2.0 Release Checklist

mhatrevi, 2025-09-03:
### Current Test Failures

ROM Intergration Tests
- [ ] test_datavault_pmp_enforcement
- [ ] test_derive_stable_key

FMC Intergration Tests
- [ ] test_boot_status_reporting

Runtime Intergration Tests
 - [ ] test_activate_firmware::test_activate_fw_id_not_in_manifest
 - [ ] test_activate_firmware::test_activate_invalid_fw_id
 - [ ] test_activate_firmware::test_activate_mcu_fw_success
 - [ ] test_activate_firmware::test_activate_mcu_soc_fw_success
 - [ ] test_activate_firmware::test_activate_soc_fw_success
 - [ ] test_activate_firmware::test_invalid_exec_bit_in_manifest
 - [ ] test_authorize_and_stash::test_authorize_from_load_address
 - [ ] test_authorize_and_stash::test_authorize_from_load_address_incorrect_digest
 - [ ] test_authorize_and_stash::test_authorize_from_staging_address
 - [ ] test_authorize_and_stash::test_authorize_from_staging_address_incorrect_digest
 - [ ] test_fe_programming::test_fe_programming_cmd
 - [ ] test_fips::test_fips_shutdown
 - [ ] test_lms::test_lms_verify_cmd
 - [ ] test_lms::test_lms_verify_failure
 - [ ] test_lms::test_lms_verify_invalid_key_lms_type
 - [ ] test_lms::test_lms_verify_invalid_lmots_type
 - [ ] test_lms::test_lms_verify_invalid_sig_lms_type

Driver Integration Tests
- [ ] test_dma_sha384

mhatrevi, 2025-09-03:
- [ ] Asserts to ensure no-cfi & cfi are not mixed.

swenson, 2025-09-03:
test_dma_sha384 is a subsystem-only test since it requires DMA, but should be fixed in https://github.com/chipsalliance/caliptra-sw/pull/2435

clundin25, 2025-09-03:
I'm seeing a lot of untracked failures in the ROM integration tests: https://github.com/chipsalliance/caliptra-sw/pull/2433, e.g. `test_image_validation::test_header_verify_vendor_lms_sig_mismatch` and more

james000, 2025-09-19:
What are the remaining items to have an official Caliptra Core 2.0.x release?

mhatrevi, 2025-09-25:
@james000 

We are working through validating the tests not ticked above on an FPGA. These are the gating items for Caliptra Core ROM 2.0 release.

clundin25, 2025-10-01:
Needs: https://github.com/chipsalliance/caliptra-sw/issues/2564

mhatrevi, 2025-11-13:
This is out of date/ The failing tests are now tracked in https://github.com/chipsalliance/caliptra-sw/issues/2214

closed 2025-11-13 by mhatrevi