chipsalliance/caliptra-sw #2441: Do we have a mechanism defined for public key hash revocation in the field?

RaunakGu, 2025-09-04:
Is there a defined mechanism for public key hash revocation in the field?
If so, could you clarify how it's triggered and managed post-deployment?

RaunakGu, 2025-09-30:
Some more details for reference, Should be better have revocation mechanism one as mentioned in "Hardware Secure Boot" OCP Doc.

**Hardware Secure Boot:**
 
[image]

**OCP Requirement:**

[image]

zhalvorsen, 2026-04-08:
We are working on a generic example implementation for revocation in MCU currently.

@daeyoungkim-87 are you looking for some form of proof of possession check?