chipsalliance/caliptra-sw #2503: UDS or FE provisioning using runtime firmware via MAILBOX commands ?

RaunakGu, 2025-09-16:
* Caliptra Version(2.1) 
* Caliptra Mode (Subsystem Mode)

hi @mhatrevi,

Is it feasible to perform UDS or FE provisioning using runtime firmware via MAILBOX commands, instead of relying on Caliptra ROM or external interfaces like JTAG TAP?

Thanks in advance

ArchanaMadhavan, 2025-09-24:
Hey @mhatrevi,  Going over the fe programming sequence, I see that it doesn't directly align with the following text from the caliptra spec.

[](https://github.com/chipsalliance/Caliptra/blob/main/doc/Caliptra.md#fuse-programming)

_"SoC shall support a field entropy programming API. The API shall support retrieving an input value from an external interface. It should cryptographically mix that value with the output of an on-die TRNG to generate the field entropy value. The API implementation shall burn the field entropy value into the first available field entropy fuse slot (or fail if no slots are available). Caliptra is expected to be in any security state."_

The implementation treats the 32 bytes as 4 slots, and the appropriate slot being passed by MCU RT. 
Also no input from the external interface is mixed in.

So largely is this completely up to SoC implementation, specifically in passive mode, including no. of slots supported ?

mhatrevi, 2025-10-02:
Fix for "Caliptra is expected to be in any security state." incorrect wording.

RaunakGu, 2026-09-01:
UDS programming using runtime Authorized command proposal is rejected in this -> https://github.com/chipsalliance/Caliptra/issues/727

so, closing this one

cc @mhatrevi

closed 2026-09-01 by RaunakGu